MPC-Berkeley/barc #13: package 'data_service' not found and Arduino issue

hchaozhe, 2017-01-05:
Hi,
    I just finished assembling the nice car following your guide. It is awesome! Thank you!
    When I try to test the car, at the section where I am supposed to type
> rosrun data_service service.py

It shows that  

> [rospack] Error: package 'data_service' not found

Do you know what is the issue here?
I tried to search online and it seems that it is with version issue. 
If so I have confronted similar issues when connecting to Arduino I thinks. When I type 

>rosrun rosserial_python serial_node.py

It shows the following after a while
>[ERROR] [WallTime: 1483592854.700273] Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

Did I do anything wrong? Or can you tell me what is the issue here? I have done a git pull before I did all above.

Thank you in advance!

jgonzal13, 2017-01-22:
Hi,
I just updated the rebuild_system script. Can you pull from master?
then open another terminal and run the command `rebuild_system`
it should rebuild all the packages, then you should be able to find data_service

hchaozhe, 2017-01-22:
Hi Jon,
     Thank you for your help!

     I did try this. I `git pull` under ~/barc folder and then do a `rebuild_system`  right after opening the second terminal. I think now it can find data_service. Also, the message 
>bash: /home/odroid/barc/workspace/devel/setup.bash: No such file or directory

disappear and now every time I open a terminal, it is normal.

But I got some messages saying that it
> Traceback (most recent call last):
  File "/home/odroid/barc/workspace/src/data_service/scripts/service.py", line 28, in <module>
    from data_service.srv import *
ImportError: No module named srv

Here is a screenshot 

[image]

In the meantime I still have the Arduino communication issue, the same message still pops up.When I type
>rosrun rosserial_python serial_node.py

It shows the following after a while

>[ERROR] [WallTime: 1483592854.700273] Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

Maybe this is still a version issue? 

Thank you in advance!

jgonzal13, 2017-02-03:
hmmm, 
1. are you running this code on the master branch? is the branch up-to-date
```
git branch
git checkout master 
git pull
```
I'm not sure why you have that issue
Are you using the odroid image from the link we provided in the documentation?
On your machine, make the packages are under the [workspace/src](https://github.com/MPC-Berkeley/barc/tree/master/workspace/src) folder (e.g. barc, barc_gui, data_service, etc.) 
Then try running catkin_make or rebulid_system again

2. I've seen that error in the past, try disconnectinf the arduino from the RC, and try running it with only the odroid and arduino connect, maybe one of the ports is causing issues. Or perhaps something is connected incorrectly
Also, perhaps the arduino is damaged (we also had that happen in the past)

hchaozhe, 2017-02-26:
Hi Jon,
      Thank you for the reply. 
For 1:

I think I tried a couple of times. I pulled and updated the folder and I checked the files. Under workspace/src, I can find everything you mentioned. The image I have is for sure the one I got from the website. As the assembly instruction suggested, I did the check the MD5 hash value to make sure the file I downloaded was correct. 
     I did what you suggested. This time when I rebuild, there is a long process, which is different from what I see before. This time I can proceed to the third step, as you can see, when I do

> bash ~/barc/scripts/register_cloud.sh

Something else is missing. I am not sure what does that mean. 

I also try to proceed to the next few step, it seems that the GUI I open shows nothing, an empty window. I think this may due to the failure in the previous step.

[image]

2) for the Arduino unit. I also tried a couple of times on different port. I also tried on my other computers to make sure the Arduino is not damaged. So far I have the difficulty on other machines as well so I am buying another one and give it another try. 

Thank you any way!

jgonzal13, 2017-02-28:
it seems python does not have the boto3 package

open a terminal, and try `pip install boto3` (actually, do this for any package that is missing)
let me know if that resolves the issue

hchaozhe, 2017-03-06:
Hi Jon,
     Thank you very much! I think it is working now! I can see the screen as the assembly instruction. One quick question I have is for the cloud registration, do I have to do this every time I run? Or I just need to do it once for one day run?
     
   
For the arduino part. I have two Arduino nano now, both of them works since I can upload several examples in the library, and they all can be uploaded to both Arduino.  I tried only connecting the arduino to the odroid and unplug all the port with camera, ultrasound etc. The same error still appear for both issues. 

[image]

Do you think it could be a version issue? If it was a port issue, can I write some code and try to detect it?

By the way, in the arduino [schematics](https://github.com/MPC-Berkeley/barc/tree/master/arduino) 
There are two ports I am not sure where to connect, the ESC D10 and Servo for D11. I can only find the throttle and steering from the RC receiver. This is not mentioned in assembly instruction. 

And I think there might be something missing in the files I should have still. When I test the camera, I am not sure whether I launch the right camera.launch file. What I run was the sensor test and the barc_cam. 

I got a warning from both the GUI and terminal. 

[image]

[image]

I also run the arduino test from here and I got the same error message in the GUI, which I think is due to the same issue. 

Do you think this could be due to the rqt_launch issue mentioned at the end of assembly instruction?

Thank you in advance for suggestion/solutions/comments of any kind!

hchaozhe, 2017-03-30:
I think on a potential issue is that I don't have the head file "EnableInterrupt.h" in the library, which is called by the arduino ino file . 

In the arduino markdown it is written that 

>Encoders now use software interrupts (formerly hardware interrupts. Install the necessary library by running source ~/barc/scripts/install_enable_interrupt.sh

I tried but I don't have this "install_enable_interrupt.sh" in scripts either. I  `git pull` before my action and I cannot find this file still.

jgonzal13, 2017-05-11:
Actually, sorry for the confusion, I realize we included code to install that script inside the install_apps.sh script.
If you open the file and look inside, you should see the commands inside there
can you try running 
` source ~/barc/scripts/install_apps.sh `
that should install the enable interrupts library for the arduino

D7 and D8 are for replaying remote control commands (steering and throttle) to the Arduino.
This is useful, for example, if you are manually controlling the RC with a remote control, but you want to modify the user inputs. 
We transmit the user's input control signals, captured by the receiver unit, to arduino ports D7 and D8.
I haven't developed any programs for this yet, but it should be useful functionality
I'll add some pictures to the documentation

Also, without trying to run a program, can you run the following
```
roscore
rosrun rosserial_python serial_node.py
```
This should run only the ROS serial communication node with between the odroid and the arduino.
Do you still get the same output?

closed 2020-12-03 by zhu-edward