MPC-Berkeley/barc #15: DimensionMismatch error in MPC_models.jl

alexbuyval, 2017-02-19:
Hi,

I am trying to run the LMPC with the simulator by the following command:

`roslaunch barc barc_sim.launch`

However, I faced with the following run-time error:

```
ERROR: LoadError: DimensionMismatch("dimensions must match")
 in promote_shape(::Tuple{Int64,Int64}, ::Tuple{Int64}) at ./operators.jl:365
 in setvalue(::Array{JuMP.Variable,1}, ::Array{Float64,2}) at /home/alex/.julia/v0.5/JuMP/src/JuMP.jl:902
 in MpcModel_pF(::MpcParams, ::ModelParams, ::TrackCoeff) at /home/alex/catkin_ws/src/barc/src/barc_lib/LMPC/MPC_models.jl:342
```

I have replaced  [line 342](https://github.com/MPC-Berkeley/barc/blob/LMPC/workspace/src/barc/src/barc_lib/LMPC/MPC_models.jl#L342) on several lines:

```
            setvalue(z_Ol[i,1],(i-1)*dt*v_ref)
            setvalue(z_Ol[i,2],0)
            setvalue(z_Ol[i,3],0)
            setvalue(z_Ol[i,4],v_ref)
            setvalue(z_Ol[i,5],0)
```

Also I have replaced [line 345](https://github.com/MPC-Berkeley/barc/blob/LMPC/workspace/src/barc/src/barc_lib/LMPC/MPC_models.jl#L345) on several lines.

After replacing there is no the error, but I am not sure that it is correct way, because I see in log that IPOPT cannot find solution:

```
WARNING: Not solved to optimality, status: UserLimit
Finished solve 1: UserLimit
WARNING: Ipopt finished with status Restoration_Failed
WARNING: Not solved to optimality, status: Error
Finished solve 2: Error
```

What is right way to solve this issue?

PS. I am using the LMPC branch, Ubuntu 14.04 and ROS Jade.

Alex

urosolia, 2017-02-20:
Hi Alex,

Which version of Julia are you using? 
We developed the simulator using Julia 0.4.7 and unfortunately the code does not run with Julia 0.5.

I tested the code on my machine using 0.4.7 and it worked correctly.

Please let me know if this fixes the issue or we need to look at it more carefully

Thanks
Ugo

alexbuyval, 2017-02-20:
Hi Ugo,

Yes, I used the Julia 0.5. I will install 0.4.7 and test it.

Thank you!

Alex

urosolia, 2017-02-20:
Hi Alex,

Let me know if you have any question about the code or the LMPC

Thanks
Ugo

alexbuyval, 2017-02-20:
I have installed the Julia 0.4.8 and the issue disappeared.

closed 2017-02-20 by alexbuyval