MPC-Berkeley/barc #17: Issue in the MPC running in Julia

angshumannitb, 2017-02-27:
[WARN] [WallTime: 1488216871.156393] Could not process inbound connection: topic types do not match: [barc/ECU] vs. [barc/ECU_raw]{'message_definition': 'float32 motor_pwm\nfloat32 servo_pwm\n\n', 'callerid': '/state_estimation', 'tcp_nodelay': '0', 'md5sum': 'aafe42831e6e4c0b016ddb7c396c7072', 'topic': '/ecu', 'type': 'barc/ECU'}

I am getting this error on running the MPC. I have made some changes on the .msg files but have done catkin_make after making the changes.

jgonzal13, 2017-02-28:
Did you modify the [CMakeLists.txt](https://github.com/MPC-Berkeley/barc/blob/master/workspace/src/barc/CMakeLists.txt#L47-54) file also?

In general, you define the msg file, add it to the CMakeLists.txt file, and then build

angshumannitb, 2017-02-28:
Yes i did that. The topics are publishing properly.

jgonzal13, 2017-03-02:
Hmmm, did you rebuild the arduino libraries that use your new ROS message?
In the rebuild_system.sh script, I use the follow set of commands to (re)generate the header files that arduino uses to publish/subscribe to topics with custom messages
```
# rebuild arduino libraries
cd $HOME/sketchbook/libraries
rm -rf ros_lib
rosrun rosserial_arduino make_libraries.py .
cd
```

angshumannitb, 2017-03-03:
Thanks Jon for the help. I had done that. I was making a mistake in the subscription for the state_estimate as I hadn't updated the message definition in the file. It fine now.

angshumannitb, 2017-03-16:
[](#!/usr/bin/env julia

#=
 Licensing Information: You are free to use or extend these projects for 
 education or research purposes provided that (1) you retain this notice
 and (2) you provide clear attribution to UC Berkeley, including a link 
 to http://barc-project.com

 Attibution Information: The barc project ROS code-base was developed
 at UC Berkeley in the Model Predictive Control (MPC) lab by Jon Gonzales
 (jon.gonzales@berkeley.edu). The cloud services integation with ROS was developed
 by Kiet Lam  (kiet.lam@berkeley.edu). The web-server app Dator was 
 based on an open source project by Bruce Wootton

 The original code is being modified by Angshuman Goswami from Clemson University
 working in Efficient Mobility via Connectivity and Control (EMC2) lab
=# 

using RobotOS
@rosimport barc.msg: ECU_raw, Encoder, Ultrasound_xy, Z_KinBkMdl
@rosimport data_service.msg: TimeData
@rosimport geometry_msgs.msg: Vector3
rostypegen()
using barc.msg
using data_service.msg
using geometry_msgs.msg
using JuMP
using Ipopt
using AmplNLWriter
#using NLopt
using geometry_msgs.msg

# define model parameters
L_a     = 0.125         # distance from CoG to front axel
L_b     = 0.125         # distance from CoG to rear axel
dt      = 0.1                # time step of system
Ff      = 0.1711        # friction coefficient
a0     = 0.1308        # air drag coeff
a       =  0.125          # distance from CoG to rear axel
b       =  0.125         # distance from CoG to front axel
I_z    =  0.24            # moment of Inertia
m     =  1.98             # mass of the vehicle

# Tire Properties
TM_F        = [25, 1.5, -0.981]
TM_R        = [25, 1.5, -1.4715]
(B1,C1,D1)  = TM_F
(B2,C2,D2)  = TM_R

# preview horizon
N       = 5

# define targets [generic values]
x_ref   = 5
y_ref   = 5

# define decision variables 
# states: position (x,y), velocity (v_x, v_y), yaw angle (phi) , yaw rate(r) 
# inputs: acceleration, steering angle 
println("Creating kinematic bicycle model ....")
mdl     = Model(solver = IpoptSolver(print_level=3))
#mdl     = Model(solver = BonminNLSolver())
#mdl     = Model(solver = NLoptSolver(algorithm=:LD_MMA))
@defVar( mdl, x[1:(N+1)] )
@defVar( mdl, y[1:(N+1)] )
@defVar( mdl, phi[1:(N+1)] )
@defVar( mdl, v_x[1:(N+1)] )
@defVar( mdl, v_y[1:(N+1)] )
@defVar( mdl, r[1:N] )
@defVar( mdl, FxR[1:N] )
[... 93 lines cut ...]
        =#
        r_opt = getValue(r[1])
        #if  -0.001<r_opt<0.001
	#      d_f_opt = 0
        #end
        phi_opt = getValue(phi[1])
        
        
        counter += 1
        if counter > 75
            # publish commands
              
              cmd = ECU_raw(a_opt, r_opt,phi_opt)
              publish(pub, cmd)
        end
        rossleep(loop_rate)
    end
end

if ! isinteractive()
    main()
end
)

I am trying to run this MPC but it is giving me out of bounds error.

angshumannitb, 2017-04-06:
I am trying to do string compare and have numeric outputs for different status like 1 for infeasible but I am getting errors.

tic()
status = solve(mdl)
toc()
ismatch(status,"Infeasible")
if ismatch(status,"Infeasible")
status = 1
println("Veh cannot move")
end

I am getting the following error:
ERROR: LoadError: MethodError: ismatch has no method matching ismatch(::Symbol, ::ASCIIString)
Closest candidates are:
ismatch(!Matched::Regex, ::AbstractString)
ismatch(!Matched::Regex, ::AbstractString, !Matched::Integer)

angshumannitb, 2017-04-06:
if( status == :Infeasible ) this solution worked

closed 2020-12-03 by zhu-edward