MPC-Berkeley/barc #21: [about input signal] I don't know how to calculate input force.

min-chuir-Park, 2017-08-11:
in the low level code, 
----------------------------------------------------------------
def pwm_converter_callback(msg): 
31     global motor_pwm, servo_pwm, b0 
33     # translate from SI units in vehicle model 
34     # to pwm angle units (i.e. to send command signal to actuators) 
36     # compute servo command 
37     str_ang     = 180.0/pi*msg.servo  # convert steering angle to degrees 
38     servo_pwm   = 92.0558 + 1.8194*str_ang  - 0.0104*str_ang**2 
40     # compute motor command 
41     FxR         =  float(msg.motor)  
42     if FxR == 0: 
43         motor_pwm = 90 
44     elif FxR > 0: 
45         motor_pwm   =  FxR/b0 + 95 
46     else: 
47         motor_pwm = 90 
48     update_arduino() 
-----------------------------------------------------------------

i don't know 
1. servo_pwm   = 92.0558 + 1.8194*str_ang  - 0.0104*str_ang**2  
2.      motor_pwm   =  FxR/b0 + 95 

how to calculate this equation.

jgonzal13, 2017-08-17:
hi @min-chuir-Park, this problem falls under model identification. 
please check out this [master's thesis](https://github.com/maxb91/Master-thesis) from one of our former contributors, Max. 
In chapter 4, "Implementation", he talks about how about model selection and fitting

The numbers in the low-level code differ from the numbers he presents in his thesis, but the approach taken is the same.

min-chuir-Park, 2017-09-18:
Thank you for your reply. But   I  can't  find  any  information master's  thesis  github about implement .  Could you write  address of  information??

closed 2020-12-03 by zhu-edward