MPC-Berkeley/barc #28: Missing Arduino library

MaartenKleijwegt, 2017-10-27:
The barc specific libraries like barc/encoder.h seem to be missing

jgonzal13, 2017-10-28:
Hi @MaartenKleijwegt ,

You can generate those header files by calling the alias `rebuild_arduino_lib` (you can see the actual commands executed [here](https://github.com/MPC-Berkeley/barc/blob/master/scripts/startup.sh)

MaartenKleijwegt, 2017-10-28:
Hi @jgonzal13,

First of all thanks for answering! I'm working on a similar vehicle as yours. However, ours is based on a Teensy 3.6 without the odroid.

Therefore I was purely interested in the arduino code, however looking at the code on your github I could not find the header files they were referring to. I'm specifically interested in how you handled the encoders.

When I couldn't find them in the arduino folder I opened the Issue. However, I get the feeling the header files are only included in the odroid image. 

Am I correct on this?

jgonzal13, 2017-10-31:
The header files are auto-generated by ROS
ROS handles communication across different devices run different programs (nodes), making data exchange very simple

For the encoders, we use interrupts for event detection; each time the hall effect sensor or optical sensor detects a change in state, it increments an internal counter variable. Because we know the geometry of the wheel, as well as the time elapsed, we can estimate velocity for each wheel

dta800, 2018-09-10:
Can you just upload the header files? It would be very useful to examine it before downloading the all image.

closed 2020-12-03 by zhu-edward