MPC-Berkeley/barc #46: Issue enabling cloud services and rqt GUI plugins missing

shathush, 2018-08-20:
Hi,

I'm part of a team that is interested in building upon the BARC platform. The available resources proved to be a great asset in allowing us to complete the hardware setup, which we did using this document: [https://static1.squarespace.com/static/568ee44d1115e085239e8d76/t/5a67e26cec212de9742e8af2/1516757615540/barc+assembly+guide+v1.0.pdf](url). 

Next, we flashed the odroid with the newest available software and began the process of enabling cloud services and testing all sensors and actuators using ROS, which we are doing using this document: [https://docs.google.com/document/d/1T8O4JhUlw09ALUGPSX7DlSO7Hc7vcKl_ahBeHncMguE/edit](url)

Unfortunately, running `rosrun data_service service.py` with roscore running returns the following error:

[image]

Do you have any suggestions to troubleshoot this error? The python package boto3 is installed as well. Additionally, when trying to use `git pull` to ensure all software is up to date, we get the following error:

[image]

Next, we tried to build the workspace and run `rqt -p ~/barc/perspectives/experiment.perspective`. According to the document referenced above, we have to run `rosrun data_service service.py` again, but that returned the same error as before. It seemed like using `source ~/barc/scripts/rebuild_system.sh` to build the workspace completed with no issues however on further inspection, there seems to be an error message:

[image]

Additionally, in the GUI window resulting from the rqt command the "BARC GUI" plugin is missing. Using the Launch (experimental) plugin, the imu can be activated however none of the barc launch files appear, as seen in the following image.

[image]

I'm sorry for the long post. We are new to working with ROS and appreciate any help we can receive. If you need anything else from us, please let us know.

Thank you,
Shathushan

jgonzal13, 2018-08-21:
Hi Shathushan,

Sorry about the trouble, the platform is still a work in progress.
Regarding the git pull error, I recommend deleting the barc folder and then going a clean git clone using https. 
(I believe the barc repo inside the odroid image we uploaded used ssh credentials for get pulls / updates to the barc repo.). 
`
rm -rf ~/barc
cd ~
git clone https://github.com/MPC-Berkeley/barc.git
`
Secondly, the instructions for the cloud service are the google document are obsolete, that was for barc version 1.0.
Thanks for mentioning the issue, I'll ask my colleague if we can upload the latest instructions for the cloud service.

shathush, 2018-08-22:
Hi Jon,

Thank you for the quick and helpful response! We appreciate the work your team has done on this platform. An updated guide would definitely be very helpful.

Thank you,
Shathushan

closed 2020-12-03 by zhu-edward