MPC-Berkeley/barc #49: ROS compatibility issue

phoenixjyb, 2019-02-12:
Hi there,

Just wondering if this project is particularly tied with ROS indigo or can be used in any later version of ROS, such as Kinetic?

I tried to run the simulation launch file (ros-kinetic) but encountered with such an error message:
'
ERROR: LoadError: Traceback (most recent call last):
  File "/home/xxx/barc/workspace/src/barc/src/stateEstimator.py", line 854, in <module>
    main()
  File "/home/xxx/barc/workspace/src/barc/src/stateEstimator.py", line 48, in main
    Q_hs[0,0]  =  rospy.get_param("/state_estimator/Qx_hs") # 0.5     # x
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 465, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/msproxy.py", line 123, in __getitem__
    raise KeyError(key)
KeyError: '/state_estimator/Qx_hs'
ArgumentError: Package RobotOS not found in current path:
- Run `Pkg.add("RobotOS")` to install the RobotOS package.
'

Would you please help check if it is the ROS version issue?

Cheers,

tonyhzheng, 2019-02-12:
Hello,

We do use ROS kinetic but I believe that error is tied with Julia. Try running the "Pkg.add(RobotOS)" command in Julia.

phoenixjyb, 2019-02-13:
> Hello,
> 
> We do use ROS kinetic but I believe that error is tied with Julia. Try running the "Pkg.add(RobotOS)" command in Julia.

Thanks very much, I'm quite new to Julia.

Another issue is about barc package in Julia. When I run Pkg.status(), it reports:

→ [e9bc4a31] barc v0.0.0 #LMPC-Shuqi (barc)
┌ Warning: Some packages (indicated with a red arrow) are not downloaded, use `instantiate` to instantiate the current environment
└ @ Pkg.Display /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.0/Pkg/src/Display.jl:190

I tried Pkg.instantiate():
julia> Pkg.instantiate()
  Updating registry at `~/.julia/registries/General`
  Updating git-repo `https://github.com/JuliaRegistries/General.git`

also tried Pkg.build():
julia> Pkg.build("barc")
  Updating registry at `~/.julia/registries/General`
  Updating git-repo `https://github.com/JuliaRegistries/General.git`

This issues still remains. Would you please help enlighten me on what possibly could go wrong? 

Cheers,

closed 2020-12-03 by zhu-edward