MPC-Berkeley/barc #52: Ipopt always finished with status Invalid_Option

martial-dong, 2019-03-18:
Hello,

I ran the simulation launch, and there was a such warning:

**started core service [/rosout]
process[controller-2]: started with pid [12511]
process[visualization-3]: started with pid [12512]
process[simulator-4]: started with pid [12513]
process[state_estimator-5]: started with pid [12514]
Python: 14.400000 m , 1441
Python: 14.400000 m , 1441
Starting LMPC node.
Julia: 14.4 m, 1441
WARNING: Ipopt finished with status Invalid_Option
WARNING: Not solved to optimality, status: Error
WARNING: Ipopt finished with status Invalid_Option
WARNING: Not solved to optimality, status: Error
WARNING: Ipopt finished with status Invalid_Option
WARNING: Not solved to optimality, status: Error
WARNING: Ipopt finished with status Invalid_Option
WARNING: Not solved to optimality, status: Error
WARNING: Ipopt finished with status Invalid_Option
WARNING: Not solved to optimality, status: Error
WARNING: Ipopt finished with status Invalid_Option
WARNING: Not solved to optimality, status: Error**
........
And the car does not move.

The program was executed with default parameters. Is there something wrong with the code or the parameter values? There are so many parameters that I don't know how to configure them to have a good result.

Anyone can give me an idea? Many thanks.

pinhofst, 2020-12-03:
I am facing this issue. Have you been able to fix it?

pinhofst, 2020-12-04:
I solved this issue by installing the HSL library. Follow the steps in the [Ipopt installation guide](https://coin-or.github.io/Ipopt/INSTALL.html#DOWNLOAD_HSL). Don't forget to also follow the following steps concerning the [setup](https://coin-or.github.io/Ipopt/INSTALL.html#LINEARSOLVERLOADER).

closed 2021-06-10 by zhu-edward